erropix/ESP32_AnalogWrite #2: How to install your library??

adjian, 2019-03-07:
I just download and copy your library to ESP32 arduino library folder. like this : C:\Users\Administrator\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.1\libraries

But I can't find it in Arduino IDE. I really want to use this library. So can you tell me how to install your library? Thank you so much.

ERROPiX, 2019-03-07:
If you are using the Arduino IDE, please load the downloaded zip file as shown in the following image 
[image]

If you are using PlatformIO you can install it from the libraries repository https://platformio.org/lib/show/5819/ESP32%20AnalogWrite

adjian, 2019-03-07:
@ERROPiX I just try your solution. It worked. Thank you for your help and your library.

closed 2019-03-07 by adjian

ERROPiX, 2019-03-07:
You are welcome, please share it if you like it 👍