erropix/ESP32_AnalogWrite #3: what pins can I use?

8ram, 2019-03-27:
I thought the esp32 can use almost all pins for PWM but I made a program that needs 9 pins to analogWrite to. What pins can I use for this? Some are not working. I am using the WiFi library.

ERROPiX, 2019-03-27:
@8ram You need to search for pin out schematic of your development board model, you can use all the pins marked with **analog**, for _WROMM32_:

[image]

closed 2019-04-20 by ERROPiX

ERROPiX, 2019-04-26:
If you look at the picture above, there is 32 pins that support pwm signal, but there is only 16 channels, some pins get disable as you use other features, and some pins have signal at boot or during upload etc..

There is a good article here https://randomnerdtutorials.com/esp32-pinout-reference-gpios/ it have a lot of details