erropix/ESP32_AnalogWrite #15: Not working

De-Jeriyo, 2024-07-11:
The example code gives me this error "c:\Users\User\Documents\Arduino\libraries\ESP32_AnalogWrite\analogWrite.cpp: In function 'int analogWriteChannel(uint8_t)':
c:\Users\User\Documents\Arduino\libraries\ESP32_AnalogWrite\analogWrite.cpp:44:9: error: 'ledcSetup' was not declared in this scope
   44 |         ledcSetup(channel, _analog_write_channels[i].frequency, _analog_write_channels[i].resolution);
      |         ^~~~~~~~~
c:\Users\User\Documents\Arduino\libraries\ESP32_AnalogWrite\analogWrite.cpp:45:9: error: 'ledcAttachPin' was not declared in this scope; did you mean 'ledcAttach'?
   45 |         ledcAttachPin(pin, channel);
      |         ^~~~~~~~~~~~~
      |         ledcAttach"

dee1337, 2024-08-23:
fixed it in https://github.com/erropix/ESP32_AnalogWrite/pull/16/commits/3e0fb55129932853f87962ea11a1dbbbc8b79694

codemasters1, 2024-10-17:
OLD ver
ledcSetup(PWM_CHANNEL, PWM_FREQUENCY, PWM_RESOLUTION);
ledcAttachPin(BUZZER_PIN, PWM_CHANNEL);

NEW ver
ledcAttach(BUZZER_PIN, PWM_FREQUENCY, PWM_RESOLUTION);

closed 2025-06-05 by erropix